rytilahti/python-yeelightbt #1: Can't install in homeassistant

Guoer-xz, 2017-01-28:
17-01-28 18:49:05 ERROR (MainThread) [homeassistant.components.light] Error while setting up platform yeelight_bt
Traceback (most recent call last):
  File "/Library/Frameworks/Python.framework/Versions/3.5/lib/python3.5/site-packages/homeassistant/helpers/entity_component.py", line 151, in _async_setup_platform
    entity_platform.add_entities, discovery_info
  File "/Library/Frameworks/Python.framework/Versions/3.5/lib/python3.5/asyncio/futures.py", line 361, in __iter__
    yield self  # This tells Task to wait for completion.
  File "/Library/Frameworks/Python.framework/Versions/3.5/lib/python3.5/asyncio/tasks.py", line 296, in _wakeup
    future.result()
  File "/Library/Frameworks/Python.framework/Versions/3.5/lib/python3.5/asyncio/futures.py", line 274, in result
    raise self._exception
  File "/Library/Frameworks/Python.framework/Versions/3.5/lib/python3.5/concurrent/futures/thread.py", line 55, in run
    result = self.fn(*self.args, **self.kwargs)
  File "/Users/guoer/.homeassistant/custom_components/light/yeelight_bt.py", line 60, in setup_platform
    add_devices_callback(lights, True)  # request an update before adding
  File "/Library/Frameworks/Python.framework/Versions/3.5/lib/python3.5/site-packages/homeassistant/helpers/entity_component.py", line 300, in add_entities
    entity.update()
  File "/Users/guoer/.homeassistant/custom_components/light/yeelight_bt.py", line 163, in update
    with self._dev:
  File "/Users/guoer/.homeassistant/custom_components/light/yeelight_bt.py", line 131, in _dev
    from yeelightbt import Lamp
ImportError: No module named 'yeelightbt'

rytilahti, 2017-01-28:
You need to download and install the library first into your environment with pip, e.g., like with the following command:
```
pip install git+https://github.com/rytilahti/python-yeelightbt.git
```

Guoer-xz, 2017-01-28:
Thanks！My English is very poor,Sorry to have troubled you.I install this in (osx 10.12.2) used sudo
However, it is still wrong。

 Downloading bluepy-1.0.5.tar.gz (210kB)
    100% |████████████████████████████████| 215kB 583kB/s 
Installing collected packages: bluepy, python-yeelightbt
  Running setup.py install for bluepy ... error
    Complete output from command /Library/Frameworks/Python.framework/Versions/3.5/bin/python3.5 -u -c "import setuptools, tokenize;__file__='/private/tmp/pip-build-0kazljo3/bluepy/setup.py';f=getattr(tokenize, 'open', open)(__file__);code=f.read().replace('\r\n', '\n');f.close();exec(compile(code, __file__, 'exec'))" install --record /tmp/pip-tsdo1p6n-record/install-record.txt --single-version-externally-managed --compile:
    running install
    Working dir is /private/tmp/pip-build-0kazljo3/bluepy
    execute make -C ./bluepy clean
    execute make -C bluepy
    Failed to compile bluepy-helper. Exiting install.
    Command was 'make -C bluepy' in /private/tmp/pip-build-0kazljo3/bluepy
    Return code was 2
    Output was:
    b"tar xzf bluez-src.tgz\ntouch ./bluez-5.29/lib/bluetooth.c ./bluez-5.29/lib/hci.c ./bluez-5.29/lib/sdp.c ./bluez-5.29/lib/uuid.c ./bluez-5.29/attrib/att.c ./bluez-5.29/attrib/gatt.c ./bluez-5.29/attrib/gattrib.c ./bluez-5.29/attrib/utils.c ./bluez-5.29/btio/btio.c ./bluez-5.29/src/log.c ./bluez-5.29/src/shared/mgmt.c ./bluez-5.29/src/shared/crypto.c ./bluez-5.29/src/shared/att.c ./bluez-5.29/src/shared/queue.c ./bluez-5.29/src/shared/util.c ./bluez-5.29/src/shared/io-glib.c ./bluez-5.29/src/shared/timeout-glib.c\ngcc -L. -Os -g -Wall -Werror -DHAVE_CONFIG_H -I./bluez-5.29/attrib -I./bluez-5.29 -I./bluez-5.29/lib -I./bluez-5.29/src -I./bluez-5.29/gdbus -I./bluez-5.29/btio -I/usr/local/Cellar/glib/2.50.1/include/glib-2.0 -I/usr/local/Cellar/glib/2.50.1/lib/glib-2.0/include -I/usr/local/opt/gettext/include -I/usr/local/Cellar/pcre/8.39/include -o bluepy-helper bluepy-helper.c ./bluez-5.29/lib/bluetooth.c ./bluez-5.29/lib/hci.c ./bluez-5.29/lib/sdp.c ./bluez-5.29/lib/uuid.c ./bluez-5.29/attrib/att.c ./bluez-5.29/attrib/gatt.c ./bluez-5.29/attrib/gattrib.c ./bluez-5.29/attrib/utils.c ./bluez-5.29/btio/btio.c ./bluez-5.29/src/log.c ./bluez-5.29/src/shared/mgmt.c ./bluez-5.29/src/shared/crypto.c ./bluez-5.29/src/shared/att.c ./bluez-5.29/src/shared/queue.c ./bluez-5.29/src/shared/util.c ./bluez-5.29/src/shared/io-glib.c ./bluez-5.29/src/shared/timeout-glib.c -L/usr/local/Cellar/glib/2.50.1/lib -L/usr/local/opt/gettext/lib -lglib-2.0 -lintl\nIn file included from bluepy-helper.c:36:\n./bluez-5.29/lib/bluetooth.h:36:10: fatal error: 'endian.h' file not found\n#include <endian.h>\n         ^\n1 error generated.\nIn file included from ./bluez-5.29/lib/bluetooth.c:38:\n./bluez-5.29/lib/bluetooth.h:36:10: fatal error: 'endian.h' file not found\n#include <endian.h>\n         ^\n1 error generated.\nIn file included from ./bluez-5.29/lib/hci.c:44:\n./bluez-5.29/lib/bluetooth.h:36:10: fatal error: 'endian.h' file not found\n#include <endian.h>\n         ^\n1 error generated.\nIn file included from ./bluez-5.29/lib/sdp.c:45:\n./bluez-5.29/lib/bluetooth.h:36:10: fatal error: 'endian.h' file not found\n#include <endian.h>\n         ^\n1 error generated.\nIn file included from ./bluez-5.29/lib/uuid.c:33:\n./bluez-5.29/lib/bluetooth.h:36:10: fatal error: 'endian.h' file not found\n#include <endian.h>\n         ^\n1 error generated.\nIn file included from ./bluez-5.29/attrib/att.c:35:\n./bluez-5.29/lib/bluetooth.h:36:10: fatal error: 'endian.h' file not found\n#include <endian.h>\n         ^\n1 error generated.\nIn file included from ./bluez-5.29/attrib/gatt.c:34:\nIn file included from ./bluez-5.29/lib/sdp.h:35:\n./bluez-5.29/lib/bluetooth/bluetooth.h:36:10: fatal error: 'endian.h' file not found\n#include <endian.h>\n         ^\n1 error generated.\nIn file included from ./bluez-5.29/attrib/gattrib.c:36:\n./bluez-5.29/lib/bluetooth.h:36:10: fatal error: 'endian.h' file not found\n#include <endian.h>\n         ^\n1 error generated.\nIn file included from ./bluez-5.29/attrib/utils.c:32:\n./bluez-5.29/lib/bluetooth.h:36:10: fatal error: 'endian.h' file not found\n#include <endian.h>\n         ^\n1 error generated.\nIn file included from ./bluez-5.29/btio/btio.c:39:\n./bluez-5.29/lib/bluetooth.h:36:10: fatal error: 'endian.h' file not found\n#include <endian.h>\n         ^\n1 error generated.\nIn file included from ./bluez-5.29/src/shared/mgmt.c:33:\n./bluez-5.29/lib/bluetooth.h:36:10: fatal error: 'endian.h' file not found\n#include <endian.h>\n         ^\n1 error generated.\nIn file included from ./bluez-5.29/src/shared/crypto.c:33:\n./bluez-5.29/src/shared/util.h:27:10: fatal error: 'byteswap.h' file not found\n#include <byteswap.h>\n         ^\n1 error generated.\nIn file included from ./bluez-5.29/src/shared/att.c:34:\n./bluez-5.29/src/shared/util.h:27:10: fatal error: 'byteswap.h' file not found\n#include <byteswap.h>\n         ^\n1 error generated.\nIn file included from ./bluez-5.29/src/shared/queue.c:28:\n./bluez-5.29/src/shared/util.h:27:10: fatal error: 'byteswap.h' file not found\n#include <byteswap.h>\n         ^\n1 error generated.\nIn file included from ./bluez-5.29/src/shared/util.c:38:\n./bluez-5.29/src/shared/util.h:27:10: fatal error: 'byteswap.h' file not found\n#include <byteswap.h>\n         ^\n1 error generated.\nmake: *** [bluepy-helper] Error 1\n"
    
    ----------------------------------------
Command "/Library/Frameworks/Python.framework/Versions/3.5/bin/python3.5 -u -c "import setuptools, tokenize;__file__='/private/tmp/pip-build-0kazljo3/bluepy/setup.py';f=getattr(tokenize, 'open', open)(__file__);code=f.read().replace('\r\n', '\n');f.close();exec(compile(code, __file__, 'exec'))" install --record /tmp/pip-tsdo1p6n-record/install-record.txt --single-version-externally-managed --compile" failed with error code 1 in /private/tmp/pip-build-0kazljo3/bluepy/

rytilahti, 2017-01-28:
Unfortunately it seems to be that bluez is not compatible with OSX (fatal error: 'byteswap.h' file not found\n#include <byteswap.h>), there is nothing this library can do about it, sorry about that.

closed 2017-01-28 by rytilahti