rytilahti/python-yeelightbt #4: yeelightbt scan

Pho7ography, 2017-02-01:
Hi,

i have install this and get this error:

pi@raspberrypi:~/python-yeelightbt $ sudo yeelightbt scan
Traceback (most recent call last):
  File "/usr/local/bin/yeelightbt", line 9, in <module>
    load_entry_point('python-yeelightbt==0.0.2', 'console_scripts', 'yeelightbt')()
  File "/usr/lib/python2.7/dist-packages/pkg_resources.py", line 356, in load_entry_point
    return get_distribution(dist).load_entry_point(group, name)
  File "/usr/lib/python2.7/dist-packages/pkg_resources.py", line 2476, in load_entry_point
    return ep.load()
  File "/usr/lib/python2.7/dist-packages/pkg_resources.py", line 2190, in load
    ['__name__'])
  File "build/bdist.linux-armv7l/egg/yeelightbt/__init__.py", line 2, in <module>
  File "build/bdist.linux-armv7l/egg/yeelightbt/lamp.py", line 5, in <module>
  File "build/bdist.linux-armv7l/egg/yeelightbt/structures.py", line 3, in <module>
ImportError: No module named construct

rytilahti, 2017-02-02:
You need to install construct with pip, I have modified the setup.py so if you reinstall it it should work. Please let me know if you have any more problems.

Pho7ography, 2017-02-02:
thanks, done the reinstall and the error is the same.

this is the log from the install:
running install
running bdist_egg
running egg_info
creating python_yeelightbt.egg-info
writing requirements to python_yeelightbt.egg-info/requires.txt
writing python_yeelightbt.egg-info/PKG-INFO
writing top-level names to python_yeelightbt.egg-info/top_level.txt
writing dependency_links to python_yeelightbt.egg-info/dependency_links.txt
writing entry points to python_yeelightbt.egg-info/entry_points.txt
writing manifest file 'python_yeelightbt.egg-info/SOURCES.txt'
reading manifest file 'python_yeelightbt.egg-info/SOURCES.txt'
writing manifest file 'python_yeelightbt.egg-info/SOURCES.txt'
installing library code to build/bdist.linux-armv7l/egg
running install_lib
running build_py
creating build
creating build/lib.linux-armv7l-2.7
creating build/lib.linux-armv7l-2.7/yeelightbt
copying yeelightbt/structures.py -> build/lib.linux-armv7l-2.7/yeelightbt
copying yeelightbt/__init__.py -> build/lib.linux-armv7l-2.7/yeelightbt
copying yeelightbt/lamp.py -> build/lib.linux-armv7l-2.7/yeelightbt
copying yeelightbt/connection.py -> build/lib.linux-armv7l-2.7/yeelightbt
copying yeelightbt/cli.py -> build/lib.linux-armv7l-2.7/yeelightbt
creating build/bdist.linux-armv7l
creating build/bdist.linux-armv7l/egg
creating build/bdist.linux-armv7l/egg/yeelightbt
copying build/lib.linux-armv7l-2.7/yeelightbt/structures.py -> build/bdist.linux-armv7l/egg/yeelightbt
copying build/lib.linux-armv7l-2.7/yeelightbt/__init__.py -> build/bdist.linux-armv7l/egg/yeelightbt
copying build/lib.linux-armv7l-2.7/yeelightbt/lamp.py -> build/bdist.linux-armv7l/egg/yeelightbt
copying build/lib.linux-armv7l-2.7/yeelightbt/connection.py -> build/bdist.linux-armv7l/egg/yeelightbt
copying build/lib.linux-armv7l-2.7/yeelightbt/cli.py -> build/bdist.linux-armv7l/egg/yeelightbt
byte-compiling build/bdist.linux-armv7l/egg/yeelightbt/structures.py to structures.pyc
byte-compiling build/bdist.linux-armv7l/egg/yeelightbt/__init__.py to __init__.pyc
byte-compiling build/bdist.linux-armv7l/egg/yeelightbt/lamp.py to lamp.pyc
byte-compiling build/bdist.linux-armv7l/egg/yeelightbt/connection.py to connection.pyc
byte-compiling build/bdist.linux-armv7l/egg/yeelightbt/cli.py to cli.pyc
creating build/bdist.linux-armv7l/egg/EGG-INFO
copying python_yeelightbt.egg-info/PKG-INFO -> build/bdist.linux-armv7l/egg/EGG-INFO
copying python_yeelightbt.egg-info/SOURCES.txt -> build/bdist.linux-armv7l/egg/EGG-INFO
copying python_yeelightbt.egg-info/dependency_links.txt -> build/bdist.linux-armv7l/egg/EGG-INFO
copying python_yeelightbt.egg-info/entry_points.txt -> build/bdist.linux-armv7l/egg/EGG-INFO
copying python_yeelightbt.egg-info/requires.txt -> build/bdist.linux-armv7l/egg/EGG-INFO
copying python_yeelightbt.egg-info/top_level.txt -> build/bdist.linux-armv7l/egg/EGG-INFO
zip_safe flag not set; analyzing archive contents...
creating dist
creating 'dist/python_yeelightbt-0.0.2-py2.7.egg' and adding 'build/bdist.linux-armv7l/egg' to it
removing 'build/bdist.linux-armv7l/egg' (and everything under it)
Processing python_yeelightbt-0.0.2-py2.7.egg
Removing /usr/local/lib/python2.7/dist-packages/python_yeelightbt-0.0.2-py2.7.egg
Copying python_yeelightbt-0.0.2-py2.7.egg to /usr/local/lib/python2.7/dist-packages
python-yeelightbt 0.0.2 is already the active version in easy-install.pth
Installing yeelightbt script to /usr/local/bin

Installed /usr/local/lib/python2.7/dist-packages/python_yeelightbt-0.0.2-py2.7.egg
Processing dependencies for python-yeelightbt==0.0.2
Searching for construct
Reading https://pypi.python.org/simple/construct/
Best match: construct 2.8.8
Downloading https://pypi.python.org/packages/b6/2c/66bab4fef920ef8caa3e180ea601475b2cbbe196255b18f1c58215940607/construct-2.8.8.tar.gz#md5=39781ff6cb36af0834c306933e2cbb3f
Processing construct-2.8.8.tar.gz
Writing /tmp/easy_install-8lgFcV/construct-2.8.8/setup.cfg
Running construct-2.8.8/setup.py -q bdist_egg --dist-dir /tmp/easy_install-8lgFcV/construct-2.8.8/egg-dist-tmp-i6t5ZZ
zip_safe flag not set; analyzing archive contents...
Adding construct 2.8.8 to easy-install.pth file

Installed /usr/local/lib/python2.7/dist-packages/construct-2.8.8-py2.7.egg
Searching for bluepy==1.0.5
Best match: bluepy 1.0.5
Processing bluepy-1.0.5-py2.7.egg
bluepy 1.0.5 is already the active version in easy-install.pth
Installing blescan script to /usr/local/bin
Installing sensortag script to /usr/local/bin

rytilahti, 2017-02-02:
Ah, I have never tested it with python2. Try to install from the beginning with python3.

Pho7ography, 2017-02-02:
Done.

error log:
pi@raspberrypi:~/python-yeelightbt $ yeelightbt scan
Traceback (most recent call last):
  File "/usr/local/bin/yeelightbt", line 9, in <module>
    load_entry_point('python-yeelightbt==0.0.2', 'console_scripts', 'yeelightbt')()
  File "/usr/lib/python3/dist-packages/pkg_resources.py", line 356, in load_entry_point
    return get_distribution(dist).load_entry_point(group, name)
  File "/usr/lib/python3/dist-packages/pkg_resources.py", line 2476, in load_entry_point
    return ep.load()
  File "/usr/lib/python3/dist-packages/pkg_resources.py", line 2190, in load
    ['__name__'])
  File "/usr/local/lib/python3.4/dist-packages/python_yeelightbt-0.0.2-py3.4.egg/yeelightbt/cli.py", line 4, in <module>
ImportError: No module named 'click'

rytilahti, 2017-02-02:
Okay, so another missing module. I'll add it to the dependencies, but you can just use pip to install it.
No module named 'click' means that you are missing click.

Pho7ography, 2017-02-02:
click installed, now different error:
pi@raspberrypi:~/python-yeelightbt $ yeelightbt scan
Scanning for 5 seconds
Traceback (most recent call last):
  File "/usr/local/bin/yeelightbt", line 9, in <module>
    load_entry_point('python-yeelightbt==0.0.2', 'console_scripts', 'yeelightbt')()
  File "/usr/local/lib/python3.4/dist-packages/click/core.py", line 722, in __call__
    return self.main(*args, **kwargs)
  File "/usr/local/lib/python3.4/dist-packages/click/core.py", line 697, in main
    rv = self.invoke(ctx)
  File "/usr/local/lib/python3.4/dist-packages/click/core.py", line 1066, in invoke
    return _process_result(sub_ctx.command.invoke(sub_ctx))
  File "/usr/local/lib/python3.4/dist-packages/click/core.py", line 895, in invoke
    return ctx.invoke(self.callback, **ctx.params)
  File "/usr/local/lib/python3.4/dist-packages/click/core.py", line 535, in invoke
    return callback(*args, **kwargs)
  File "/usr/local/lib/python3.4/dist-packages/python_yeelightbt-0.0.2-py3.4.egg/yeelightbt/cli.py", line 58, in scan
  File "/usr/local/lib/python3.4/dist-packages/bluepy-1.0.5-py3.4.egg/bluepy/btle.py", line 631, in scan
    self.start()
  File "/usr/local/lib/python3.4/dist-packages/bluepy-1.0.5-py3.4.egg/bluepy/btle.py", line 568, in start
    self._startHelper(iface=self.iface)
  File "/usr/local/lib/python3.4/dist-packages/bluepy-1.0.5-py3.4.egg/bluepy/btle.py", line 210, in _startHelper
    universal_newlines=True)
  File "/usr/lib/python3.4/subprocess.py", line 859, in __init__
    restore_signals, start_new_session)
  File "/usr/lib/python3.4/subprocess.py", line 1457, in _execute_child
    raise child_exception_type(errno_num, err_msg)
FileNotFoundError: [Errno 2] No such file or directory: '/usr/local/lib/python3.4/dist-packages/bluepy-1.0.5-py3.4.egg/bluepy/bluepy-helper'

rytilahti, 2017-02-02:
Okay, that's because of broken bluepy-helper, you need to go to /usr/local/lib/python3.4/dist-packages/bluepy-1.0.5-py3.4.egg/bluepy/ and run 'make' there. You may need to install the compiler and other tools by doing this:
sudo apt-get install build-essential libglib2.0-dev

if the make doesn't succeed directly.

Then run this to give the helper access to bluetooth:
setcap cap_net_admin,cap_net_raw+eip /usr/local/lib/python3.4/dist-packages/bluepy-1.0.5-py3.4.egg/bluepy/bluepy-helper

Thanks for testing, I'll update the documentation at some point.

Pho7ography, 2017-02-02:
working, thank
Now i gone try to config in HA.

deennoo, 2017-02-02:
thx for upate, all working now except scan 

```
Scanning for 5 seconds
Traceback (most recent call last):
  File "/usr/local/bin/yeelightbt", line 9, in <module>
    load_entry_point('python-yeelightbt==0.0.2', 'console_scripts', 'yeelightbt')()
  File "/usr/local/lib/python3.4/dist-packages/click-6.7-py3.4.egg/click/core.py", line 722, in __call__
    return self.main(*args, **kwargs)
  File "/usr/local/lib/python3.4/dist-packages/click-6.7-py3.4.egg/click/core.py", line 697, in main
    rv = self.invoke(ctx)
  File "/usr/local/lib/python3.4/dist-packages/click-6.7-py3.4.egg/click/core.py", line 1066, in invoke
    return _process_result(sub_ctx.command.invoke(sub_ctx))
  File "/usr/local/lib/python3.4/dist-packages/click-6.7-py3.4.egg/click/core.py", line 895, in invoke
    return ctx.invoke(self.callback, **ctx.params)
  File "/usr/local/lib/python3.4/dist-packages/click-6.7-py3.4.egg/click/core.py", line 535, in invoke
    return callback(*args, **kwargs)
  File "/usr/local/lib/python3.4/dist-packages/python_yeelightbt-0.0.2-py3.4.egg/yeelightbt/cli.py", line 58, in scan
  File "/usr/local/lib/python3.4/dist-packages/bluepy-1.0.5-py3.4.egg/bluepy/btle.py", line 631, in scan
    self.start()
  File "/usr/local/lib/python3.4/dist-packages/bluepy-1.0.5-py3.4.egg/bluepy/btle.py", line 576, in start
    self._mgmtCmd("scanend")
  File "/usr/local/lib/python3.4/dist-packages/bluepy-1.0.5-py3.4.egg/bluepy/btle.py", line 240, in _mgmtCmd
    "Failed to execute mgmt cmd '%s'" % (cmd))
bluepy.btle.BTLEException: Failed to execute mgmt cmd 'scanend'
```

rytilahti, 2017-02-02:
Sorry, there isn't much I can do if bluepy's scanning is not working, you'll have to find another way to get the mac address, so I'm closing this now.

closed 2017-02-02 by rytilahti

Pho7ography, 2017-02-03:
Thanks for the help ;)

jenccnej, 2017-02-13:
Hello
I'm getting strange similar errors on scan
```
Scanning for 5 seconds
Traceback (most recent call last):
  File "/usr/local/bin/yeelightbt", line 9, in <module>
    load_entry_point('python-yeelightbt==0.0.2', 'console_scripts', 'yeelightbt')()
  File "/usr/local/lib/python2.7/dist-packages/click/core.py", line 722, in __call__
    return self.main(*args, **kwargs)
  File "/usr/local/lib/python2.7/dist-packages/click/core.py", line 697, in main
    rv = self.invoke(ctx)
  File "/usr/local/lib/python2.7/dist-packages/click/core.py", line 1066, in invoke
    return _process_result(sub_ctx.command.invoke(sub_ctx))
  File "/usr/local/lib/python2.7/dist-packages/click/core.py", line 895, in invoke
    return ctx.invoke(self.callback, **ctx.params)
  File "/usr/local/lib/python2.7/dist-packages/click/core.py", line 535, in invoke
    return callback(*args, **kwargs)
  File "/usr/local/lib/python2.7/dist-packages/yeelightbt/cli.py", line 58, in scan
    devs = scan.scan(sec)
  File "/usr/local/lib/python2.7/dist-packages/bluepy-1.0.5-py2.7.egg/bluepy/btle.py", line 631, in scan
    self.start()
  File "/usr/local/lib/python2.7/dist-packages/bluepy-1.0.5-py2.7.egg/bluepy/btle.py", line 568, in start
    self._startHelper(iface=self.iface)
  File "/usr/local/lib/python2.7/dist-packages/bluepy-1.0.5-py2.7.egg/bluepy/btle.py", line 210, in _startHelper
    universal_newlines=True)
  File "/usr/lib/python2.7/subprocess.py", line 710, in __init__
    errread, errwrite)
  File "/usr/lib/python2.7/subprocess.py", line 1335, in _execute_child
    raise child_exception
OSError: [Errno 2] No such file or directory
```