rytilahti/python-yeelightbt #5: Ha plugin installation consulting

Guoer-xz, 2017-02-04:
Again, I've got the MAC address from other sources, is it still possible to complete the previous steps to make the HA plug-in work, or just fill in the correct yaml can be completed. I had a question about the OSX installation plugin before

rytilahti, 2017-02-04:
The plugin requires the same library as the client tool, which isn't unfortunately available on OSX. Are you running HA on some Linux-based device? If not, I can't really help at this point.

Guoer-xz, 2017-02-04:
Thank you for your answer, I finally want to use on the homekit, before the device is also used HB or ha transfer to the HB way, it seems that you are a very good tool and I missed:(

closed 2017-02-04 by Guoer-xz