rytilahti/python-yeelightbt #6: Ha plugin installation error

Pho7ography, 2017-02-06:
Hi again,

I try to install the custom component in HA, but this is the error, can't you give me some light ?

17-02-05 19:14:38 ERROR (MainThread) [homeassistant.loader] Unable to find component light.yeelight_bt
17-02-05 19:14:38 ERROR (MainThread) [homeassistant.bootstrap] Unable to find platform light.yeelight_bt

rytilahti, 2017-02-11:
How have you installed it? You need to put it in ~/.homeassistant/custom_components/light directory, and then you can use light.yeelight_bt in your config.

Pho7ography, 2017-02-11:
I have done that and validate that homeassistant user have rights in the directory.

Pho7ography, 2017-02-12:
Hi,
I have  change de directory and now:
  File "/home/homeassistant/.homeassistant/custom_components/light/yeelight_bt.py", line 59, in setup_platform
    add_devices_callback(lights, True)  # request an update before adding
  File "/srv/homeassistant/lib/python3.4/site-packages/homeassistant/helpers/entity_component.py", line 300, in add_entities
    entity.update()
  File "/home/homeassistant/.homeassistant/custom_components/light/yeelight_bt.py", line 162, in update
    with self._dev:
  File "/home/homeassistant/.homeassistant/custom_components/light/yeelight_bt.py", line 130, in _dev
    from yeelightbt import Lamp
ImportError: No module named 'yeelightbt'

rytilahti, 2017-03-08:
Have you installed the module itself? You can do that by running "pip install ." in the source directory.

rytilahti, 2017-05-05:
Closing for inactivity, this sounds like missing installation.

closed 2017-05-05 by rytilahti