rytilahti/python-yeelightbt #11: state-request works but control doesn't work

andy2301, 2017-06-26:
I followed exactly the steps and the state request returns correct information. However, the control doesn't work. 

Each time I set the color, I can see that the lamp sets the color to my requested one, however it reverts back to the original state right away. The same happens to setting temperature and setting brightness. 

I also tried turning on and off, but neither worked.

The following is the debug output.
`$ yeelightbt -d color 0 0 255
DEBUG:yeelightbt.connection:Trying to connect to F8:24:41:E3:32:E6
DEBUG:yeelightbt.connection:Connected to F8:24:41:E3:32:E6
DEBUG:yeelightbt.connection:Writing b'0100' to 22 with with_response=False
DEBUG:yeelightbt.lamp:>> {'type': 'Pair'} (wait: 0.5)
DEBUG:yeelightbt.connection:Writing b'436702000000000000000000000000000000' to 18 with with_response=False
DEBUG:yeelightbt.connection:Got notification from 21: b'436301000000000000000000000000000000'
DEBUG:yeelightbt.lamp:<< b'436301000000000000000000000000000000'
DEBUG:yeelightbt.lamp:pairing res: Container:
    type = PairingResult (total 13)
    response = None
    payload = Container:
        pairing_status = NotPaired (total 9)
DEBUG:yeelightbt.lamp:>> {'type': 'GetState'} (wait: 0.5)
DEBUG:yeelightbt.connection:Writing b'434400000000000000000000000000000000' to 18 with with_response=False
DEBUG:yeelightbt.connection:Got notification from 21: b'43450101f200ff000100001a000000000000'
DEBUG:yeelightbt.lamp:<< b'43450101f200ff000100001a000000000000'
Setting color: 0 0 255
DEBUG:yeelightbt.lamp:>> {'green': 0, 'red': 0, 'blue': 255, 'type': 'SetColor'} (wait: 0.5)
DEBUG:yeelightbt.connection:Writing b'43410000ff00000000000000000000000000' to 18 with with_response=False
`
`$ yeelightbt -d off
DEBUG:yeelightbt.connection:Trying to connect to F8:24:41:E3:32:E6
DEBUG:yeelightbt.connection:Connected to F8:24:41:E3:32:E6
DEBUG:yeelightbt.connection:Writing b'0100' to 22 with with_response=False
DEBUG:yeelightbt.lamp:>> {'type': 'Pair'} (wait: 0.5)
DEBUG:yeelightbt.connection:Writing b'436702000000000000000000000000000000' to 18 with with_response=False
DEBUG:yeelightbt.connection:Got notification from 21: b'436301000000000000000000000000000000'
DEBUG:yeelightbt.lamp:<< b'436301000000000000000000000000000000'
DEBUG:yeelightbt.lamp:pairing res: Container:
    type = PairingResult (total 13)
    response = None
    payload = Container:
        pairing_status = NotPaired (total 9)
DEBUG:yeelightbt.lamp:>> {'type': 'GetState'} (wait: 0.5)
DEBUG:yeelightbt.connection:Writing b'434400000000000000000000000000000000' to 18 with with_response=False
DEBUG:yeelightbt.connection:Got notification from 21: b'43450101f200ff0001000019000000000000'
DEBUG:yeelightbt.lamp:<< b'43450101f200ff0001000019000000000000'
DEBUG:yeelightbt.lamp:>> {'type': 'SetOnOff', 'state': False} (wait: 0.5)
DEBUG:yeelightbt.connection:Writing b'434002000000000000000000000000000000' to 18 with with_response=False
`

andy2301, 2017-06-26:
I'm running the official Raspian Jessie on my Raspberry Pi 3.

Note that, the following command doesn't seem to work.

**pi@raspi3:~ $ setcap cap_net_admin,cap_net_raw+eip bluepy-helper**
_unable to set CAP_SETFCAP effective capability: Operation not permitted_

rytilahti, 2017-06-26:
You need to run setcap as root (e.g. with sudo) for it to work, as said by the error message (Operation not permitted).

From the sound of it, I think your lamp is in a mode where it itself adjusts the settings and will directly override the setting. If so there's not much the library can do (without trying to deactivate such modes).