rytilahti/python-yeelightbt #12: Hass io installation

ziat007, 2017-08-16:
Could a hass io repository be added for simple (no root access required) installation?

Without this I don't think it is easy to install this component. (Hass io only grants access to a virtual environment with home assistant preinstalled)

bachulator, 2017-11-10:
Yes integration with hassio would be super cool. We need a plug-in for this extension I assume.

fernandex00, 2018-02-04:
Some news about this?

lbischof, 2018-09-02:
Wouldn't the `custom deps deployment` addon work?

Duveaux, 2019-01-07:
I know this is old, but you can install it in hass.io using the following config for `custom deps deployment`.

```
{
  "pypi": [
    "git+https://github.com/rytilahti/python-yeelightbt/",
    "construct==2.9.25"
  ],
  "apk": []
}
```

matteos1, 2019-02-08:
> I know this is old, but you can install it in hass.io using the following config for `custom deps deployment`.
> 
> ```
> {
>   "pypi": [
>     "git+https://github.com/rytilahti/python-yeelightbt/",
>     "construct==2.9.25"
>   ],
>   "apk": []
> }
> ```

could you explain how is possible install in hassio?

Duveaux, 2019-02-14:
> > I know this is old, but you can install it in hass.io using the following config for `custom deps deployment`.
> > ```
> > {
> >   "pypi": [
> >     "git+https://github.com/rytilahti/python-yeelightbt/",
> >     "construct==2.9.25"
> >   ],
> >   "apk": []
> > }
> > ```
> 
> could you explain how is possible install in hassio?

What exactly is unclear? In hass.io you have to install the `custom deps deployment` add-on. You can use that as the configuration for the add-on. Run it once and you should be done.

lasry1, 2019-02-14:
I've tried to installed through the custom deps deployment add-on and the lamp never bloody connects

Got exception: cannot import name 'SymmetricMapping'

rytilahti, 2019-02-26:
The problem with symmetricmapping should be fixed by #28, the custom component may or may not work + it will also capture the device for the runtime of homeassistant (which can be problematic if your bluetooth adapter does not support multiple connections, or if you want to control it outside homeassistant) and will thus also drain the battery (in case of candelas).

If someone is interested in working to make it more usable, I can give some help if needed, but I'm not personally going to put much effort on this project.

rytilahti, 2019-06-08:
Closing as stale.

closed 2019-06-08 by rytilahti