rytilahti/python-yeelightbt #13: Installation Problem

ReiVaX1508, 2018-01-20:
Hi;

Thanks for your work; but for me doesn't work and i've got any problems

Simple command doesn't work !

`pi@raspberry:~ $ yeelightbt scan
Traceback (most recent call last):
  File "/usr/local/bin/yeelightbt", line 11, in <module>
    load_entry_point('python-yeelightbt==0.0.2', 'console_scripts', 'yeelightbt')()
  File "/usr/lib/python2.7/dist-packages/pkg_resources/__init__.py", line 561, in load_entry_point
    return get_distribution(dist).load_entry_point(group, name)
  File "/usr/lib/python2.7/dist-packages/pkg_resources/__init__.py", line 2631, in load_entry_point
    return ep.load()
  File "/usr/lib/python2.7/dist-packages/pkg_resources/__init__.py", line 2291, in load
    return self.resolve()
  File "/usr/lib/python2.7/dist-packages/pkg_resources/__init__.py", line 2297, in resolve
    module = __import__(self.module_name, fromlist=['__name__'], level=0)
  File "/usr/local/lib/python2.7/dist-packages/yeelightbt/__init__.py", line 2, in <module>
    from .lamp import Lamp
  File "/usr/local/lib/python2.7/dist-packages/yeelightbt/lamp.py", line 5, in <module>
    from .structures import Request, Response
  File "/usr/local/lib/python2.7/dist-packages/yeelightbt/structures.py", line 70, in <module>
    "unknown" / Const(Byte, 0x00)
  File "/usr/local/lib/python2.7/dist-packages/construct/core.py", line 1277, in __init__
    super(Const, self).__init__(subcon)
  File "/usr/local/lib/python2.7/dist-packages/construct/core.py", line 287, in __init__
    raise TypeError("subcon should be a Construct field")
TypeError: subcon should be a Construct field`

Have you got solutions for ????

rytilahti, 2018-01-20:
This is caused by a change in construct, to fix this the parameter order for those `Const`s should be swapped. (So in this case on line 70, it should be `Const(0x00, Byte)`)

ReiVaX1508, 2018-01-20:
I apply this change but same problem ??

pi@raspberry $ yeelightbt
Traceback (most recent call last):
  File "/usr/local/bin/yeelightbt", line 11, in <module>
    load_entry_point('python-yeelightbt==0.0.2', 'console_scripts', 'yeelightbt'               )()
  File "/usr/lib/python2.7/dist-packages/pkg_resources/__init__.py", line 561, i               n load_entry_point
    return get_distribution(dist).load_entry_point(group, name)
  File "/usr/lib/python2.7/dist-packages/pkg_resources/__init__.py", line 2631,                in load_entry_point
    return ep.load()
  File "/usr/lib/python2.7/dist-packages/pkg_resources/__init__.py", line 2291,                in load
    return self.resolve()
  File "/usr/lib/python2.7/dist-packages/pkg_resources/__init__.py", line 2297,                in resolve
    module = __import__(self.module_name, fromlist=['__name__'], level=0)
  File "/usr/local/lib/python2.7/dist-packages/yeelightbt/__init__.py", line 2,                in <module>
    from .lamp import Lamp
  File "/usr/local/lib/python2.7/dist-packages/yeelightbt/lamp.py", line 5, in <               module>
    from .structures import Request, Response
  File "/usr/local/lib/python2.7/dist-packages/yeelightbt/structures.py", line 1               62, in <module>
    "dev_uid" / Const(Byte, 0x02))
  File "/usr/local/lib/python2.7/dist-packages/construct/core.py", line 1277, in                __init__
    super(Const, self).__init__(subcon)
  File "/usr/local/lib/python2.7/dist-packages/construct/core.py", line 287, in                __init__
    raise TypeError("subcon should be a Construct field")
TypeError: subcon should be a Construct field

rytilahti, 2018-01-20:
That's coming from a different line number, you have to change all the consts accordingly. (just swap the order)

ReiVaX1508, 2018-01-20:
It's OK for me; tanks for solution

But now; impossible to contol my yeelight:

ERROR:yeelightbt.lamp:got exception, tries left 3: no mapping for 6, no default either
    (parsing) -> msg -> payload -> status -> pairing_status
ERROR:yeelightbt.lamp:got exception, tries left 3: no mapping for 3, no default either
    (parsing) -> msg -> payload -> status -> pairing_status
ERROR:yeelightbt.lamp:got exception, tries left 3: no mapping for 3, no default either
    (parsing) -> msg -> payload -> status -> pairing_status

rytilahti, 2018-01-20:
What kind of lamp it is? I'm currently doing some changes to how it all works (to add support for Candelas), it could be that this problem will also get fixed with it.

ReiVaX1508, 2018-01-21:
My lamp is a Yeelight Bedlamp in first bluetooth version.

ReiVaX1508, 2018-02-04:
Thanks for this new version. But i've got another problem when i try to install librairies:

 Cloning https://github.com/rytilahti/python-yeelightbt/ to /tmp/pip-2s929_ez-build
    Complete output from command python setup.py egg_info:
    Traceback (most recent call last):
      File "<string>", line 1, in <module>
      File "/tmp/pip-2s929_ez-build/setup.py", line 9
        description='Python library for interfacing with yeelight's bt lights',
                                                                  ^
    SyntaxError: invalid syntax

    ----------------------------------------
Command "python setup.py egg_info" failed with error code 1 in /tmp/pip-2s929_ez-build/

rytilahti, 2018-02-04:
Oops, sorry, that is fixed now.

ReiVaX1508, 2018-02-04:
For install it's ok and thanks for this, but when i try command ON, get this:

pi@AIOLIA:~ $ yeelightbt on
INFO:yeelightbt.connection:Requesting characteristics for uuid 8f65073d-9f57-4aaa-afea-397d19d5bbeb
INFO:yeelightbt.connection:Requesting characteristics for uuid aa7d3f34-2d4f-41e0-807f-52fbf8cf7443
Container:
    type = PairingResult (total 13)
    pairing_status = UnknownState (total 12)
ERROR:yeelightbt.lamp:got exception on {'type': 'GetState'}, tries left 3: Device disconnected
INFO:yeelightbt.connection:Requesting characteristics for uuid 8f65073d-9f57-4aaa-afea-397d19d5bbeb
INFO:yeelightbt.connection:Requesting characteristics for uuid aa7d3f34-2d4f-41e0-807f-52fbf8cf7443
Container:
    type = PairingResult (total 13)
    pairing_status = UnknownState (total 12)
ERROR:yeelightbt.lamp:got exception on {'type': 'GetState'}, tries left 2: Device disconnected
INFO:yeelightbt.connection:Requesting characteristics for uuid 8f65073d-9f57-4aaa-afea-397d19d5bbeb
INFO:yeelightbt.connection:Requesting characteristics for uuid aa7d3f34-2d4f-41e0-807f-52fbf8cf7443
Container:
    type = PairingResult (total 13)
    pairing_status = UnknownState (total 12)
ERROR:yeelightbt.lamp:got exception on {'type': 'GetState'}, tries left 1: Device disconnected
INFO:yeelightbt.connection:Requesting characteristics for uuid 8f65073d-9f57-4aaa-afea-397d19d5bbeb
INFO:yeelightbt.connection:Requesting characteristics for uuid aa7d3f34-2d4f-41e0-807f-52fbf8cf7443
Container:
    type = PairingResult (total 13)
    pairing_status = UnknownState (total 12)
Traceback (most recent call last):
  File "/usr/local/bin/yeelightbt", line 11, in <module>
    load_entry_point('python-yeelightbt==0.0.3.1', 'console_scripts', 'yeelightbt')()
  File "/home/pi/.local/lib/python3.5/site-packages/click/core.py", line 722, in __call__
    return self.main(*args, **kwargs)
  File "/home/pi/.local/lib/python3.5/site-packages/click/core.py", line 697, in main
    rv = self.invoke(ctx)
  File "/home/pi/.local/lib/python3.5/site-packages/click/core.py", line 1063, in invoke
    Command.invoke(self, ctx)
  File "/home/pi/.local/lib/python3.5/site-packages/click/core.py", line 895, in invoke
    return ctx.invoke(self.callback, **ctx.params)
  File "/home/pi/.local/lib/python3.5/site-packages/click/core.py", line 535, in invoke
    return callback(*args, **kwargs)
  File "/home/pi/.local/lib/python3.5/site-packages/click/decorators.py", line 17, in new_func
    return f(get_current_context(), *args, **kwargs)
  File "/usr/local/lib/python3.5/dist-packages/yeelightbt/cli.py", line 59, in cli
    lamp.state()
  File "/usr/local/lib/python3.5/dist-packages/yeelightbt/lamp.py", line 45, in _wrap
    raise _ex
  File "/usr/local/lib/python3.5/dist-packages/yeelightbt/lamp.py", line 35, in _wrap
    self._conn.wait(wait)
  File "/usr/local/lib/python3.5/dist-packages/yeelightbt/connection.py", line 74, in wait
    self._conn.waitForNotifications(timeout=0.1)
  File "/home/pi/.local/lib/python3.5/site-packages/bluepy/btle.py", line 516, in waitForNotifications
    resp = self._getResp(['ntfy','ind'], timeout)
  File "/home/pi/.local/lib/python3.5/site-packages/bluepy/btle.py", line 369, in _getResp
    resp = self._waitResp(wantType + ['ntfy', 'ind'], timeout)
  File "/home/pi/.local/lib/python3.5/site-packages/bluepy/btle.py", line 326, in _waitResp
    raise BTLEException(BTLEException.DISCONNECTED, "Device disconnected")
bluepy.btle.BTLEException: Device disconnected

rytilahti, 2018-02-05:
Is the light on initially? Do you see it "breathing"? If yes, that's the time where you need to press the button like instructed in the README.

ReiVaX1508, 2018-02-05:
Thank for this. It's OK and working. Great Work.

ReiVaX1508, 2018-02-05:
Just a little things..... i configure any commands for yeelight in domoticz; but when i push an action, the effect on the lamp is so long.... ~10seconds.

Have you got same problem ??

rytilahti, 2018-02-14:
With an effect you mean how long it takes from the lamp to respond? It can take a while indeed, as the newest version looks specifically for which handles to use for communication, I think I'll add an optimization for this soon.

On the other hand it also depends on your hardware, if other devices are using the bt and so on. Here it takes some 4-5 seconds to react to those commands now. Are you using the library interface directly or are you just executing the `yeelightbt` command? If the former, you can simply call `connect()` only once and keep the connection open until it is not needed anymore. 

Btw, be wary of upgrading construct, version 2.9.26 is the last one which is still working and I'm not sure when I'll have some time to fix this.