rytilahti/python-yeelightbt #15: from Construct developer, a note

arekbulski, 2018-02-14:
I am the developer of Construct (a parsing library) since I took over the project 2 years ago. Please refer any other projects that also use this library to this post, or copy it over to them. 

Construct is undergoing heavy changes at the moment, similar to those between 2.5->2.8, including addition of compiler feature (to make it much faster than Kaitai), but also several classes are going to be removed (for a good reason), and also few were added. API will be unstable for about a month.

What you can do:
- All scheduled work is posted as tickets, in [Issues](https://github.com/construct/construct/issues) and also on [Kanban](https://github.com/construct/construct/projects/3) page. If you subscribe (watch repository) then you will have advance notice of any incoming changes. Unfortunately those tickets also include minor work items, so you would get some clutter notifications too. I appologise for that.
- I encourage everyone to browse those tickets, vote for feature requests, but also post objections for planned removal of features, or changes in semantics. Subscribing therefore is recommended.
- You can find what was already changed (at any moment) on [Transition page](https://construct.readthedocs.io/en/latest/transition29.html), which is like a growing changelog. It gets updated whenever something gets added or removed, but since the order is not chronological, new items do not end up on top or bottom of the list.
- You can find currently available classes and semantics on revamped [Core API pages](https://construct.readthedocs.io/en/latest/index.html#api-reference). Those pages are very extensive and always uptodate with the implementation.
- **Feel free, at any point in time, to open a new Issue and ask questions or make complaints** about breakage. At minimum, I can offer advice and possible alternatives. People very often use the classes in ways that were not designed for, like use String classes to process Bytes.
- **Feel free, at any point in time, to open a new Issue and request changes.** If you need a new class, and make a good clear case, I would be more than willing to add it to core library to support your project.
- API should be stable within a month, so either do version pinning and deal with it then, or keep fixing your code at every release (which is ~3 day cycle at the moment).

Close this topic at your discretion.

arekbulski, 2018-02-14:
Somewhat related to #13 which was Const changing arguments order.

arekbulski, 2018-02-14:
I briefly reviewed your code and found a problem: Embedded(Switch) is not supported anymore, but there is EmbeddedSwitch macro that simulates that. You may have some trouble wrapping your head around how it works. See docs:
https://construct.readthedocs.io/en/latest/api/conditional.html#construct.EmbeddedSwitch

arekbulski, 2018-02-14:
Could you grant me write access so I could work on patches on local branches?

rytilahti, 2018-02-14:
I will take a look into it at some point later, I suppose. I feel that the current switch structure this project is using is kind of burdensome, but at this point I don't have time to put into reworking it. PRs for fixing issues are more than welcome :-)

closed 2018-02-17 by arekbulski