rytilahti/python-yeelightbt #17: Issues (different) on both PY2.7 and PY3.5

Bob2345de, 2018-02-22:
Install seems successful but error as follows installed on PY2.7.9:
^Croot@raspberrypi:~# export YEELIGHTBT_MAC=F8:24:41:C0:36:AF
root@raspberrypi:~# yeelightbt

Traceback (most recent call last):
  File "/usr/local/bin/yeelightbt", line 11, in <module>
    load_entry_point('python-yeelightbt==0.0.3.1', 'console_scripts', 'yeelightbt')()
  File "/usr/lib/python2.7/dist-packages/pkg_resources/__init__.py", line 561, in load_entry_point
    return get_distribution(dist).load_entry_point(group, name)
  File "/usr/lib/python2.7/dist-packages/pkg_resources/__init__.py", line 2631, in load_entry_point
    return ep.load()
  File "/usr/lib/python2.7/dist-packages/pkg_resources/__init__.py", line 2291, in load
    return self.resolve()
  File "/usr/lib/python2.7/dist-packages/pkg_resources/__init__.py", line 2297, in resolve
    module = __import__(self.module_name, fromlist=['__name__'], level=0)
  File "/usr/local/lib/python2.7/dist-packages/yeelightbt/__init__.py", line 2, in <module>
    from .lamp import Lamp
  File "/usr/local/lib/python2.7/dist-packages/yeelightbt/lamp.py", line 192
    def set_temperature(self, kelvin: int, brightness: int):
                                                     ^
SyntaxError: invalid syntax

**And on PY3.5:**
root@raspberrypi:~# export YEELIGHTBT_MAC=F8:24:41:C0:36:AF
root@raspberrypi:~# yeelightbt --help
Traceback (most recent call last):
  File "/usr/local/bin/yeelightbt", line 11, in <module>
    load_entry_point('python-yeelightbt==0.0.3.1', 'console_scripts', 'yeelightbt')()
  File "/usr/lib/python3/dist-packages/pkg_resources/__init__.py", line 561, in load_entry_point
    return get_distribution(dist).load_entry_point(group, name)
  File "/usr/lib/python3/dist-packages/pkg_resources/__init__.py", line 2631, in load_entry_point
    return ep.load()
  File "/usr/lib/python3/dist-packages/pkg_resources/__init__.py", line 2291, in load
    return self.resolve()
  File "/usr/lib/python3/dist-packages/pkg_resources/__init__.py", line 2297, in resolve
    module = __import__(self.module_name, fromlist=['__name__'], level=0)
  File "/usr/local/lib/python3.5/dist-packages/yeelightbt/__init__.py", line 2, in <module>
    from .lamp import Lamp
  File "/usr/local/lib/python3.5/dist-packages/yeelightbt/lamp.py", line 7, in <module>
    from .structures import Request, Response, StateResult
  File "/usr/local/lib/python3.5/dist-packages/yeelightbt/structures.py", line 4, in <module>
    from construct import (Struct, Int8ub, Int16ub, Const, Padded, Byte, Enum, Bytes, If, FlagsEnum,
ImportError: cannot import name 'SymmetricMapping'

Bob2345de, 2018-02-22:
Found the issue. There were some changes made to "construct" as described here: https://github.com/construct/construct/issues/642

Overwriting the constructs directory with the contents from version 2.9.25 (available here https://github.com/construct/construct/releases/tag/v2.9.25 ) solved the issue on Python3.5, it now works! 

I suspect that some changes will need to be made to yeelightbt to keep it compatible going forward.

Thanks

rytilahti, 2018-02-23:
The construct's API is currently unstable, but this issue will probably be fixed by PR #16. Btw, you should use pip to install the packages instead of manually copying files over, so `pip install construct==2.9.25` should work :-)

matteos1, 2018-06-25:
python-yeelightbt requires Python '>=3.4' but the running Python is 2.7.13 not work install

arekbulski, 2018-07-21:
Construct has achieved what you would call a "stable API".

rytilahti, 2018-07-21:
:+1: I will close this issue, this project does not support python 2, and the symmetricmapping problem is tracked in #18.

closed 2018-07-21 by rytilahti