rytilahti/python-yeelightbt #18: Cannot import name 'SymmetricMapping'

t871631092, 2018-03-09:
2018-03-09 14:57:14 ERROR (Thread-19) [custom_components.light.yeelight_bt] Got exception: cannot import name 'SymmetricMapping'
2018-03-09 14:57:14 INFO (MainThread) [homeassistant.core] Bus:Handling <Event system_log_event[L]: source=custom_components/light/yeelight_bt.py, exception=, level=ERROR, timestamp=1520607434.851715, message=Got exception: cannot import name 'SymmetricMapping'>
## 
I can‘t find the SymmetricMapping by anyway，I'm crazy

hcoohb, 2018-03-12:
I have the same error.
Is it because of construct 2.9.37 ?

t871631092, 2018-03-12:
I have installed the 2.9.25  Should i upgrade ？

t871631092, 2018-03-14:
it work on my first time run hass --open-ui
then the second time i run 
it error again

hcoohb, 2018-04-03:
Construct is now pinned at 2.9.41 on home assistant.

But yeelightbt is still showing the same error.
(And the code from the Pull Request 16 is not working either)
Anyone could point us in the right direction ?

frittes, 2018-06-09:
Any chance to solve this issue in the near future?

dan9550, 2018-07-12:
Same for me. Even if you downgrade if you're using home assistant this will break other components.

arekbulski, 2018-07-21:
In some recent Construct version, SymmetricMapping was renamed to Mapping. This error is caused by using an outdated version.

rytilahti, 2018-07-21:
The fix is being worked in #19, but it's a bit slow process as I'm busy with other things currently. If someone wants to step up and finish that porting, it'd be more than welcome.

onlyyou, 2018-11-11:
hey,
did make commande
pip install construct==2.9.25
but still have error
pi@raspberrypi:~ $ yeelightbt scan
Traceback (most recent call last):
  File "/usr/local/bin/yeelightbt", line 11, in <module>
    load_entry_point('python-yeelightbt==0.0.3.1', 'console_scripts', 'yeelightbt')()
  File "/home/pi/.local/lib/python3.5/site-packages/pkg_resources/__init__.py", line 484, in load_entry_point
    return get_distribution(dist).load_entry_point(group, name)
  File "/home/pi/.local/lib/python3.5/site-packages/pkg_resources/__init__.py", line 2725, in load_entry_point
    return ep.load()
  File "/home/pi/.local/lib/python3.5/site-packages/pkg_resources/__init__.py", line 2343, in load
    return self.resolve()
  File "/home/pi/.local/lib/python3.5/site-packages/pkg_resources/__init__.py", line 2349, in resolve
    module = __import__(self.module_name, fromlist=['__name__'], level=0)
  File "/home/pi/.local/lib/python3.5/site-packages/yeelightbt/__init__.py", line 2, in <module>
    from .lamp import Lamp
  File "/home/pi/.local/lib/python3.5/site-packages/yeelightbt/lamp.py", line 7, in <module>
    from .structures import Request, Response, StateResult
  File "/home/pi/.local/lib/python3.5/site-packages/yeelightbt/structures.py", line 4, in <module>
    from construct import (Struct, Int8ub, Int16ub, Const, Padded, Byte, Enum, Bytes, If, FlagsEnum,
ImportError: cannot import name 'SymmetricMapping'

thanks

lasry1, 2019-02-11:
I have the same problem, i've installed the plugin with custom deps deployment (i'm running Hassio with HA 0.87.0) but i can't make the light available...

Thieum64, 2019-02-26:
Hi, 
It seems to be a good script to control Candela but I'm facing the same issue 'ImportError: cannot import name 'SymmetricMapping' when trying to install it. 
Is there a way to workarround ? 
Thanks

rytilahti, 2019-02-26:
So I just merged the changes and the code is where I left it long time ago. The basic functionality should work by using the command line tool for what it's worth, maybe someone finds this useful and creates PRs to fix other issues, I have no personal interest on working further on this.

closed 2019-02-26 by rytilahti