rytilahti/python-yeelightbt #21: Question about light temperature

pszafer, 2018-09-04:
Hi,

I have Xiaomi Bedside lamp and your library after set colour is always returning "temperature 0".
If I set temp by app it returns proper light temp, at least looks like K.
Is it some different unit or why is it 0?

rytilahti, 2018-09-04:
First or second gen lamp? It could be a bug, but IIRC it also reports the kelvin values properly when the lamp itself is in the temperature mode, and otherwise simply 0.

pszafer, 2018-09-04:
Bluetooth only, so first generation I think.
I'm asking because for some reason google_assistant component in HA is trying to get light temperature, and if it is then I've got division by 0 error.

I've changed it to somethink like this for now in _status_cb:
```
        else:
            if self._dev.temperature > 0:
                self._ct = int(kelvin_to_mired(self._dev.temperature))
            else:
                self._ct = int(kelvin_to_mired(2400))
```

but for me it somehow no logical that I got temperature 0, it should always return some K temp in any light mode.

rytilahti, 2019-06-08:
Please feel to submit a PR to change the behavior, if you wish to do so. But returning "some K temp" in any mode would cause inconsistent behavior when changing between modes / depending on what commands are being executed, so I don't think that's the correct approach.