rytilahti/python-yeelightbt #23: yeelightbt got error, is it due to phyton version2?

hamwong, 2018-09-12:
I am lake of using phyton, please kindly advise

[14:46:21] openhabian@OpenhabianPi:/usr/local/bin$ yeelightbt
Traceback (most recent call last):
  File "/usr/local/bin/yeelightbt", line 9, in <module>
    load_entry_point('python-yeelightbt==0.0.3.1', 'console_scripts', 'yeelightbt')()
  File "/usr/lib/python2.7/dist-packages/pkg_resources.py", line 356, in load_entry_point
    return get_distribution(dist).load_entry_point(group, name)
  File "/usr/lib/python2.7/dist-packages/pkg_resources.py", line 2476, in load_entry_point
    return ep.load()
  File "/usr/lib/python2.7/dist-packages/pkg_resources.py", line 2190, in load
    ['__name__'])
  File "/usr/local/lib/python2.7/dist-packages/yeelightbt/__init__.py", line 2, in <module>
    from .lamp import Lamp
  File "/usr/local/lib/python2.7/dist-packages/yeelightbt/lamp.py", line 192
    def set_temperature(self, kelvin: int, brightness: int):
                                    ^
SyntaxError: invalid syntax

hamwong, 2018-09-12:
my Python version Python 2.7.9 and Python 3.4.2

rytilahti, 2018-09-12:
Yes, this library only supports python 3.4+. I'm unsure how you have been able to install it even on python2, as that should not be possible.

closed 2018-09-12 by rytilahti

rytilahti, 2018-10-05:
You have to reinstall it for python3, now it's looking for the package under /usr/lib/python3, but I assume you have it only installed for python2.

hamwong, 2018-10-07:
Hi, I did as you said uninstall the yeelightbt on 2.7 and reinstall on 3 as below and still got error after,
I try it on another raspberry which didn't install yeelightbt before, result still same, all on python 3.4.2

sudo pip3 install git+https://github.com/rytilahti/python-yeelightbt/
[sudo] password for openhabian:
Downloading/unpacking git+https://github.com/rytilahti/python-yeelightbt/
  Cloning https://github.com/rytilahti/python-yeelightbt/ to /tmp/pip-yk5_83cw-build
  Running setup.py (path:/tmp/pip-yk5_83cw-build/setup.py) egg_info for package from git+https://github.com/rytilahti/python-yeelightbt/
    /usr/lib/python3.4/distutils/dist.py:260: UserWarning: Unknown distribution option: 'python_requires'
      warnings.warn(msg)

Downloading/unpacking bluepy (from python-yeelightbt==0.0.3.1)
  Downloading bluepy-1.2.0.tar.gz (509kB): 509kB downloaded
  Running setup.py (path:/tmp/pip-build-m6h62321/bluepy/setup.py) egg_info for package bluepy

Downloading/unpacking construct (from python-yeelightbt==0.0.3.1)
  Downloading construct-2.9.45.tar.gz (56kB): 56kB downloaded
  Running setup.py (path:/tmp/pip-build-m6h62321/construct/setup.py) egg_info for package construct

Downloading/unpacking click (from python-yeelightbt==0.0.3.1)
  Downloading Click-7.0-py2.py3-none-any.whl (81kB): 81kB downloaded
Installing collected packages: bluepy, construct, click, python-yeelightbt
  Running setup.py install for bluepy
    Working dir is /tmp/pip-build-m6h62321/bluepy
    execute make -C ./bluepy clean
    execute make -C bluepy -j1

    Installing blescan script to /usr/local/bin
    Installing sensortag script to /usr/local/bin
    Installing thingy52 script to /usr/local/bin
  Running setup.py install for construct

  Running setup.py install for python-yeelightbt
    /usr/lib/python3.4/distutils/dist.py:260: UserWarning: Unknown distribution option: 'python_requires'
      warnings.warn(msg)

    Installing yeelightbt script to /usr/local/bin
Successfully installed bluepy construct click python-yeelightbt
Cleaning up...

[21:53:17] openhabian@OpenhabianPi-Server:~$ yeelightbt --help
Traceback (most recent call last):
  File "/usr/local/bin/yeelightbt", line 9, in <module>
    load_entry_point('python-yeelightbt==0.0.3.1', 'console_scripts', 'yeelightbt')()
  File "/usr/lib/python3/dist-packages/pkg_resources.py", line 356, in load_entry_point
    return get_distribution(dist).load_entry_point(group, name)
  File "/usr/lib/python3/dist-packages/pkg_resources.py", line 2476, in load_entry_point
    return ep.load()
  File "/usr/lib/python3/dist-packages/pkg_resources.py", line 2190, in load
    ['__name__'])
  File "/usr/local/lib/python3.4/dist-packages/yeelightbt/__init__.py", line 2, in <module>
    from .lamp import Lamp
  File "/usr/local/lib/python3.4/dist-packages/yeelightbt/lamp.py", line 7, in <module>
    from .structures import Request, Response, StateResult
  File "/usr/local/lib/python3.4/dist-packages/yeelightbt/structures.py", line 4, in <module>
    from construct import (Struct, Int8ub, Int16ub, Const, Padded, Byte, Enum, Bytes, If, FlagsEnum,
ImportError: cannot import name 'SymmetricMapping'

hamwong, 2018-10-07:
I found issue come from construct, currently using  construct 2.9.45, after downgrade to 2.9.25, no errors and run well, don't know why

rytilahti, 2018-10-07:
Yes, this codebase hasn't been upgraded to work with newer construct versions (https://github.com/rytilahti/python-yeelightbt/issues/18).

hamwong, 2018-10-07:
Pair fail with below info, do you know why?

 sudo yeelightbt scan
Scanning for 5 seconds
Devices found:
Candela f8:24:41:c0:68:2a (yeelight_ms), rssi=-41
[01:41:13] openhabian@OpenhabianPi-Server:~$ sudo yeelightbt --mac f8:24:41:c0:68:2a
INFO:yeelightbt.connection:Requesting characteristics for uuid 8f65073d-9f57-4aaa-afea-397d19d5bbeb
INFO:yeelightbt.connection:Requesting characteristics for uuid aa7d3f34-2d4f-41e0-807f-52fbf8cf7443
Container:
    type = PairingResult (total 13)
    pairing_status = PairFailed (total 10)
Pairing failed, exiting

rytilahti, 2018-10-07:
Did you press the button when the light was flashing during the pairing process?