rytilahti/python-yeelightbt #24: The Candela pairs every etime

redviper2100, 2018-11-02:
Hey,

Nice work! 
I have a little question. I want to use your code to integrate some Candelas with the Domoticz Home Automation System.
But from the moment I send the command it take around 7 seconds for the Candela to receive it.
This is probably because the Candela doesn't stay connected all the time but it pairs every time I send a command.

Is there any way  to make it faster? If I use this in a home automation system it's kinda strange to touch a switch and eat some popcorn while the light will turn on at some moment. 

The code is very nice!

matteos1, 2019-02-08:
hi i think that in home automation there is a problem with python because the  pip install construct==2.9.41 and is not 
pip install construct==2.9.25.
there is a possibilty to start this /config/custom_components with version 2.9.25.
thank you

rytilahti, 2019-06-08:
@matteos1 that was likely fixed by #28, that is unrelated to this issue.