rytilahti/python-yeelightbt #26: Can't launch on hassbian yeelightbt

cioppz, 2018-11-27:
Installation on hassbian with pip3 command worked, but when I try to launch the yeelightbt command I get "Command not found", any idea?

rytilahti, 2018-11-27:
Maybe a confusion between virtualenvs/installations? Without more input it is not possible to say what is wrong in your setup. However, you should be aware that this library/tool is not really maintained and requires in its current state a specific version of the construct library, meaning that it won't work if you are using the same environment with other programs (e.g. homeassistant) requiring a more recent version.

cioppz, 2018-11-27:
hi @rytilahti I simply run the command for the installation from the pi user, then when I try to launch yeelightbt 'command not found'. I also think that I'll to install it inside the environment of home assistant to intrigate it, right?

rytilahti, 2019-02-26:
Integrating with homeassistant will likely require some work, but with #28 merged it should be doable to make this work in the same virtualenv where the homeassistant resides.

Controlling the lamps via the console tool should work no matter how it is installed, I'm not sure why it is not installing the control script in your case, sorry.

rytilahti, 2019-06-08:
Closing as stale.

closed 2019-06-08 by rytilahti