rytilahti/python-yeelightbt #27: object has no attribute  'update_ha_state'

rengaboy, 2019-02-25:
I have a problem with my installation. When i run yeelight from console, all is ok, i got responses and i can toggle lamp on and off.
`/home/pi/.local/bin/yeelightbt --mac f8:24:******** on
INFO:yeelightbt.connection:Requesting characteristics for uuid 8f65073d-9f57-4aaa-afea-397d19d5bbeb
INFO:yeelightbt.connection:Requesting characteristics for uuid aa7d3f34-2d4f-41e0-807f-52fbf8cf7443
Container:
    type = PairingResult (total 13)
    pairing_status = PairSuccess (total 11)
We are paired.
Container:
    type = StateResult (total 11)
    state = False
    mode = LampMode.Color
    red = 2
    green = 7
    blue = 8
    white = 0
    brightness = 0
    temperature = 0
    temp_fraction = 0
Container:
    type = StateResult (total 11)
    state = True
    mode = LampMode.Color
    red = 2
    green = 7
    blue = 8
    white = 0
    brightness = 0
    temperature = 0
    temp_fraction = 0
`
But when I,m trying to start a hass plugin, i got an error.
`2019-02-25 23:44:47 ERROR (Thread-14) [custom_components.yeelight_bt.light] Initializing f8:24:41:c0:e7:9a
2019-02-25 23:44:47 ERROR (MainThread) [homeassistant.components.device_tracker] Error setting up platform bluetooth_le_tracker
2019-02-25 23:44:49 ERROR (Thread-14) [yeelightbt.lamp] got exception on {'type': 'GetState'}, tries left 3: 'YeelightBT' object has no attribute 'update_ha_state'
2019-02-25 23:44:51 ERROR (Thread-14) [yeelightbt.lamp] got exception on {'type': 'GetState'}, tries left 2: 'YeelightBT' object has no attribute 'update_ha_state'
2019-02-25 23:44:53 ERROR (Thread-14) [yeelightbt.lamp] got exception on {'type': 'GetState'}, tries left 1: 'YeelightBT' object has no attribute 'update_ha_state'
2019-02-25 23:44:55 ERROR (MainThread) [homeassistant.components.light] yeelight_bt: Error on device update!
Traceback (most recent call last):
  File "/srv/homeassistant/lib/python3.5/site-packages/homeassistant/helpers/entity_platform.py", line 248, in _async_add_entity
    await entity.async_device_update(warning=False)
  File "/srv/homeassistant/lib/python3.5/site-packages/homeassistant/helpers/entity.py", line 349, in async_device_update
    await self.hass.async_add_executor_job(self.update)
  File "/usr/lib/python3.5/asyncio/futures.py", line 380, in __iter__
    yield self  # This tells Task to wait for completion.
  File "/usr/lib/python3.5/asyncio/tasks.py", line 304, in _wakeup
    future.result()
  File "/usr/lib/python3.5/asyncio/futures.py", line 293, in result
    raise self._exception
  File "/usr/lib/python3.5/concurrent/futures/thread.py", line 55, in run
    result = self.fn(*self.args, **self.kwargs)
  File "/home/homeassistant/.homeassistant/custom_components/yeelight_bt/light.py", line 163, in update
    self._dev.state()
  File "/srv/homeassistant/lib/python3.5/site-packages/yeelightbt/lamp.py", line 45, in _wrap
    raise _ex
  File "/srv/homeassistant/lib/python3.5/site-packages/yeelightbt/lamp.py", line 35, in _wrap
    self._conn.wait(wait)
  File "/srv/homeassistant/lib/python3.5/site-packages/yeelightbt/connection.py", line 74, in wait
    self._conn.waitForNotifications(timeout=0.1)
  File "/srv/homeassistant/lib/python3.5/site-packages/bluepy/btle.py", line 560, in waitForNotifications
    resp = self._getResp(['ntfy','ind'], timeout)
  File "/srv/homeassistant/lib/python3.5/site-packages/bluepy/btle.py", line 416, in _getResp
    self.delegate.handleNotification(hnd, data)
  File "/srv/homeassistant/lib/python3.5/site-packages/yeelightbt/connection.py", line 89, in handleNotification
    self._callbacks[handle](data)
  File "/srv/homeassistant/lib/python3.5/site-packages/yeelightbt/lamp.py", line 243, in handle_notification
    self._status_cb(self)
  File "/home/homeassistant/.homeassistant/custom_components/yeelight_bt/light.py", line 157, in _status_cb
    self.update_ha_state()
AttributeError: 'YeelightBT' object has no attribute 'update_ha_state'`

rytilahti, 2019-02-26:
The component was done so long ago that the homeassistant api has changed; to make it work some changes are required (on top of that, the version of construct library required for homeassistant and this repository clash, see #19), unfortunately there is no easy solution for this. If your light has wifi connectivity, you can use the app to enable the lan mode and simply use the `yeelight` included in homeassistant.

rengaboy, 2019-02-26:
Hello, Teemu.
Unfortunately, Xiaomi Candle has no WiFi, and not presented in MiHome application. Using of your module is the only way to manage this light.

Best Regards, Pavel Sosnovsky.
mail to: pavel.sosnovsky@gmail.com

rytilahti, 2019-02-26:
Ah, I see. You could try to replace the `self.update_ha_state()` call with a call to `self.schedule_update_ha_state()`, that should fix the above error. However, as I mentioned, the version requirements clash which requires deeper changes to the library.

rytilahti, 2019-02-26:
I just fixed that part, but looking at the homeassistant platform code, it is likely quite problematic in real use (as it will keep the connection alive forever for updates, reserving the device). As I'm not personally using this, it is unlikely that I'll be spending time to make it more usable at this point, PRs are welcome!

commit e076e719 (2019-02-26): replace deprecated update_ha_state(), fixes #27
closed 2019-02-26 by rytilahti