rytilahti/python-yeelightbt #29: New BLE Mesh devices support

pcmester, 2019-03-25:
On CES 2019, Xiaomi announced a new line of Yeelight devices, the Yeelight Mesh. Theese lamps works with BLE, and currently there is no API for control the Mesh devices.
The Yeelight Mesh devices are:
YLDP10YL
YLDP09YL
YLSD01YL 
YLSD04YL

germanotero, 2019-05-23:
I have a bunch of the YLSd01yl (downlights) and I'm trying to get to handle them with no success... I don't have the knowledge about BLE to do this.. I was able to connect with gatttool, but cant handle anything from there...

rytilahti, 2019-06-08:
PRs are welcome, the best way to figure out how to communicate with these devices is to use the official app and reverse engineer what is happening. If you want to do that, http://nilhcem.com/iot/reverse-engineering-simple-bluetooth-devices (and maybe https://www.instructables.com/id/Reverse-Engineering-Smart-Bluetooth-Low-Energy-Dev/) is a good starting point.

pcmester, 2019-06-20:
I would like to buy from theese bulbs and do the reverse engineering, but the problem is, that we can not capture the packets, because the light is not directly controlled by the phone, there is a BLE gateway between them. In this case, this is the Xiaoi AI speaker, or the Xiaomi Smart Clock.

germanotero, 2019-06-20:
yeah, I bought the clock and also have the spots, but the thing is that they are unusable even with the clock... they are slow AF, and I don't think the issue is the lap... On the other hand, the lap is pretty easy to hack, so I might hack it and change it to a mysensors node instead of dealing with this chinesse BS...

pcmester, 2019-07-16:
Could you sniff the BLE traffic? And about the smart clock, can it be controlled by python-miio?

rytilahti, 2019-07-19:
I assume the clock also has to initialize a connection to the bulbs (without knowing how the mesh system works), which can take some time.. I have never looked deeper into how fast BLE communication can be done, but my anecdotal evidence is that there's always some delay.

The smart clock is not currently supported (see https://github.com/rytilahti/python-miio/issues/505), but I have access to one so I will try to figure out how it works at some point.

rezmus, 2019-07-20:
there will be soon new gateway from yeelight to control mesh lights.

https://homekitnews.com/wp-content/uploads/2019/06/yeelight-BT-gateway.jpg

nickolik, 2019-07-23:
These lamps can be connected via gateways:
- Mi AI Alarm Clock
- Mi Smart Air Detector
- Yeelight Voice Assistant
I have Mi AI Alarm Clock and Mi Smart Air Detector.
If you set the lamp to turn on in the hub on the event (click button). This script works only when the gateway is connected to the Internet.
So to control the lamps you always need access to the Internet. I really hope that the new yeelight BT gateway will work autonomously.

primebit, 2020-05-23:
Hello! Any news on this topic?

rytilahti, 2020-05-23:
Not that I know of at least, this project is more or less dormant from my side at least, but I haven't heard anyone else working on those devices either.

Zuz666, 2020-10-21:
> but I haven't heard anyone else working on those devices either.
Now It works via: https://github.com/AlexxIT/XiaomiGateway3/

TheOnlyBeardedBeast, 2021-06-18:
I still cant work with the device YLDP001-A (RGBW) any info about this one?

rezmus, 2021-06-18:
YLDP001-A does not have mijia integration, it connects to GH directly.