rytilahti/python-yeelightbt #30: Everything works from HA for a while... then connection drops

fede87GitHub, 2019-04-12:
Hi!
This is the issue regarding Candela from HA: after restarting HA (docker container) everything works perfectly for a while. Then it seems that bluetooth connection drops and works again just restarting HA (I guess because of initialization).

This is the error:
Updating yeelight_bt light took longer than the scheduled update interval 0:00:30
5:22 PM helpers/entity_platform.py (WARNING) - message first occured at 5:04 PM and shows up 35 times

System: home assistant from a docker container (hassio) on a raspberry pi model 3+
Issue with Candela

I've seen that in the custom component keep_connection is correctly set to True.

Could you help me?
Thanks,
Federico

rytilahti, 2019-04-12:
The `keep_connection` means that the platform does not release the connection after executing a command or an update. So in case that connection breaks down as you said, it won't reconnect, which would explain the behavior you are seeing.

You can set it to False and then it will only form a connection when it is needed, going around that problem. The real fix would be probably to reconnect in case the connection gets dropped at some point. A patch to fix the behavior would be welcome, as I'm not actively using this project anymore.

rytilahti, 2019-06-08:
Closing as stale.

closed 2019-06-08 by rytilahti