rytilahti/python-yeelightbt #32: How does this script work?

netRunner0, 2019-04-25:
I successfully find the lamps through scanning. But still could not connect to them. Always timeout error. Lamps reset before connecting

netRunner0, 2019-04-25:
`pi@raspberrypi:~ $ sudo yeelightbt scan
Scanning for 5 seconds
Devices found:
Candela f8:24:41:c1:bc:4b (yeelight_ms), rssi=-72
Candela f8:24:41:c1:30:7a (yeelight_ms), rssi=-60
pi@raspberrypi:~ $ export YEELIGHTBT_MAC=f8:24:41:c1:30:7a
pi@raspberrypi:~ $ yeelightbt
INFO:yeelightbt.connection:Requesting characteristics for uuid 8f65073d-9f57-4aaa-afea-397d19d5bbeb
Traceback (most recent call last):
  File "/usr/local/bin/yeelightbt", line 11, in <module>
    load_entry_point('python-yeelightbt==0.0.4', 'console_scripts', 'yeelightbt')()
  File "/home/pi/.local/lib/python3.5/site-packages/click/core.py", line 764, in __call__
    return self.main(*args, **kwargs)
  File "/home/pi/.local/lib/python3.5/site-packages/click/core.py", line 717, in main
    rv = self.invoke(ctx)
  File "/home/pi/.local/lib/python3.5/site-packages/click/core.py", line 1114, in invoke
    return Command.invoke(self, ctx)
  File "/home/pi/.local/lib/python3.5/site-packages/click/core.py", line 956, in invoke
    return ctx.invoke(self.callback, **ctx.params)
  File "/home/pi/.local/lib/python3.5/site-packages/click/core.py", line 555, in invoke
    return callback(*args, **kwargs)
  File "/home/pi/.local/lib/python3.5/site-packages/click/decorators.py", line 17, in new_func
    return f(get_current_context(), *args, **kwargs)
  File "/home/pi/.local/lib/python3.5/site-packages/yeelightbt/cli.py", line 60, in cli
    lamp.connect()
  File "/home/pi/.local/lib/python3.5/site-packages/yeelightbt/lamp.py", line 91, in connect
    notify_char = self._conn.get_characteristics(Lamp.NOTIFY_UUID)
  File "/home/pi/.local/lib/python3.5/site-packages/yeelightbt/connection.py", line 82, in get_characteristics
    return self._conn.getCharacteristics(uuid=uuid)
  File "/home/pi/.local/lib/python3.5/site-packages/bluepy/btle.py", line 508, in getCharacteristics
    rsp = self._getResp('find')
  File "/home/pi/.local/lib/python3.5/site-packages/bluepy/btle.py", line 407, in _getResp
    resp = self._waitResp(wantType + ['ntfy', 'ind'], timeout)
  File "/home/pi/.local/lib/python3.5/site-packages/bluepy/btle.py", line 368, in _waitResp
    raise BTLEGattError("Bluetooth command failed", resp)
bluepy.btle.BTLEGattError: Bluetooth command failed (code: 14, error: Request attribute has encountered an unlikely error)
`

rytilahti, 2019-06-08:
There seems to be something wrong in your setup, I suppose. Last time (it's been a long time...) I had no troubles connecting with my candela. That error seems to happen when the library tries to ind out the necessary information to communicate with the device, so your guess why this is happening is as good as mine...