rytilahti/python-yeelightbt #33: Integration yeelight_bt not found when trying to verify its light platform

asha220, 2019-07-25:
When I try check config in Home assistant it show this error
"Integration yeelight_bt not found when trying to verify its light platform" 

I tried to place the component in custom_components/light/ and custom_components/yeelight_bt/light
but that doesn't help

Home Assistant version is 0.96.3

lasry1, 2019-08-01:
I have the same issue, and also had another one that had this issue too but they sorted out:

https://github.com/jnimmo/hass-dmx/issues/21

I'm trying to do the same with this but i can't make it work

rytilahti, 2019-08-02:
Thanks for the link @lasry1!

So my understanding is, that the folder inside `custom_components` needs to be called `yeelight_bt` with only two three files in it:
* `__init__.py` (which can be empty)
* light.py which is the code itself, and 
* `manifest.json` like https://developers.home-assistant.io/docs/en/creating_integration_manifest.html

I don't have access to any device to try this myself, so I hope someone else will figure out a solution and create a PR for it.

asha220, 2019-08-02:
Hi @rytilahti. Yes, problem was in file's structure, I change it and Home Assistant can see this component. Thanks!

closed 2019-08-02 by asha220

asha220, 2019-08-02:
Ок. Some times later. Now I have problem with change color and temp, I want fix it too.

lasry1, 2019-08-02:
Thanks @rytilahti !
I was able to add it without problems now however i can't get the light working... I don't really know what i'm doing wrong.
I'm using the latest version of HassIO

However if i use the yeelight_bt.py included in this repo (python-yeelightbt/custom_components/light/yeelight_bt.py) the light is always unavailable 

If i use the yeelight_bt.py from python-yeelightbt/custom_components/yeelight_bt/light/yeelight_bt.py then i don't get the light coming up on homeassistant