rytilahti/python-yeelightbt #34: No module named 'yeelightbt'

lasry1, 2019-08-03:
Hi,

I'm running hassio and i'm unable to make it work. After placing the Yeelight_BT.yaml file inside the custom_components and set the details into the configuration.yaml i get an error:

ERROR (SyncWorker_14) [custom_components.yeelight_bt.light] Got exception: No module named 'yeelightbt'

I can't use pip install as i'm running hassio so i'm out of ideas of how to make it work

Anybody know how to solve it?

Thanks in advance

lasry1, 2019-08-03:
If i use the file from "python-yeelightbt/custom_components/yeelight_bt/light/yeelight_bt.py"
i get the following error message 

2019-08-03 12:54:52 ERROR (MainThread) [homeassistant.components.light] yeelight_bt: Error on device update!
Traceback (most recent call last):
  File "/usr/src/homeassistant/homeassistant/helpers/entity_platform.py", line 261, in _async_add_entity
    await entity.async_device_update(warning=False)
  File "/usr/src/homeassistant/homeassistant/helpers/entity.py", line 380, in async_device_update
    await self.hass.async_add_executor_job(self.update)
  File "/usr/local/lib/python3.7/concurrent/futures/thread.py", line 57, in run
    result = self.fn(*self.args, **self.kwargs)
  File "/config/custom_components/yeelight_bt/light.py", line 163, in update
    with self._dev:
  File "/config/custom_components/yeelight_bt/light.py", line 131, in _dev
    from yeelightbt import Lamp
ModuleNotFoundError: No module named 'yeelightbt'

Calcu2006, 2019-10-13:
i'm into the same situation... I can't achieve to make it work into hassio/home assistant

rytilahti, 2020-01-22:
The homeassistant integration is very old, and likely doesn't work with current homeassistant releases without some changes. That error indicates that the library is not installed though, did you do `pip install python-yeelightbt` and are the files installed correctly under the site-packages directory?

rytilahti, 2020-04-02:
Closing as stale, the current custom component has a `manifest.json` file with the requirement for this library, so it should now be automatically installed when needed.

closed 2020-04-02 by rytilahti