rytilahti/python-yeelightbt #35: Unable to connect to device

qwertyway, 2019-12-16:
I've struggling for several days. I can successfully scan for the lamp, but when I try to connect I get

` DEBUG:connection:Trying to connect to f8:24:41:e2:18:a9
Running  /usr/local/lib/python3.7/dist-packages/bluepy/bluepy-helper
Sent:  conn f8:24:41:e2:18:a9 public hci0

Got: '# bluepy-helper.c version 1.3.0 built at 21:27:19 on Dec 15 2019\n'
Got: "rsp=$stat\x1estate=$tryconn\x1edst='f8:24:41:e2:18:a9\x1emtu=h0\x1esec='low\n"
Got: "rsp=$stat\x1estate=$disc\x1emtu=h0\x1esec='low\n"
Stopping  /usr/local/lib/python3.7/dist-packages/bluepy/bluepy-helper
WARNING:connection:Unable to connect to the device f8:24:41:e2:18:a9, retrying: Failed to connect to peripheral f8:24:41:e2:18:a9, addr type: public
Running  /usr/local/lib/python3.7/dist-packages/bluepy/bluepy-helper
Sent:  conn f8:24:41:e2:18:a9 public

Got: '# bluepy-helper.c version 1.3.0 built at 21:27:19 on Dec 15 2019\n'
Got: "rsp=$stat\x1estate=$tryconn\x1edst='f8:24:41:e2:18:a9\x1emtu=h0\x1esec='low\n"
Got: "rsp=$stat\x1estate=$disc\x1emtu=h0\x1esec='low\n"
Stopping  /usr/local/lib/python3.7/dist-packages/bluepy/bluepy-helper
ERROR:connection:Second connection try to f8:24:41:e2:18:a9 failed: Failed to connect to peripheral f8:24:41:e2:18:a9, addr type: public`

Any suggestions what can be wrong? Please, I really need this

rytilahti, 2020-01-22:
Sorry for the delay. Unfortunately it is hard to say what is wrong, for some reason the backend library cannot form a connection and that's all I can say.

qwertyway, 2020-01-22:
No problem. I've already completely disassembled the lamp, installed esp8266 in it, and connected to Home Assistant via wi-fi. Now I can control it as I wish

Here's a video for this DIY project, and the body of Xiaomi lamp perfectly fits. 
[Gyver Lamp](https://youtu.be/771-Okf0dYs)

Anyway, thank you. I'm closing the issue

closed 2020-01-22 by qwertyway

rytilahti, 2020-01-22:
Ah, I see, that looks great! Anyway, I have not tested that lamp with this library at all, only the candelas and the original bluetooth-only bedside lamp are known to work with this.