rytilahti/python-yeelightbt #38: Candela not work in HomeAssistent

Titanius, 2020-04-01:
Protokolldetails ( ERROR )
Logger: homeassistant.components.websocket_api.http.connection.2951776048
Source: custom_components/yeelight_bt/light.py:190 
Integration: websocket_api (documentation, issues) 
First occurred: 8:14:54 (1 occurrences) 
Last logged: 8:14:54

Helper not started (did you call connect()?)
Traceback (most recent call last):
  File "/usr/src/homeassistant/homeassistant/components/websocket_api/commands.py", line 134, in handle_call_service
    connection.context(msg),
  File "/usr/src/homeassistant/homeassistant/core.py", line 1234, in async_call
    await asyncio.shield(self._execute_service(handler, service_call))
  File "/usr/src/homeassistant/homeassistant/core.py", line 1257, in _execute_service
    await handler.func(service_call)
  File "/usr/src/homeassistant/homeassistant/helpers/entity_component.py", line 198, in handle_service
    self._platforms.values(), func, call, required_features
  File "/usr/src/homeassistant/homeassistant/helpers/service.py", line 402, in entity_service_call
    future.result()  # pop exception if have
  File "/usr/src/homeassistant/homeassistant/helpers/entity.py", line 599, in async_request_call
    await coro
  File "/usr/src/homeassistant/homeassistant/helpers/service.py", line 433, in _handle_entity_call
    await result
  File "/usr/src/homeassistant/homeassistant/components/light/__init__.py", line 243, in async_handle_light_on_service
    await light.async_turn_on(**params)
  File "/usr/src/homeassistant/homeassistant/helpers/entity.py", line 624, in async_turn_on
    await self.hass.async_add_job(ft.partial(self.turn_on, **kwargs))
  File "/usr/local/lib/python3.7/concurrent/futures/thread.py", line 57, in run
    result = self.fn(*self.args, **self.kwargs)
  File "/config/custom_components/yeelight_bt/light.py", line 190, in turn_on
    self._dev.turn_on()
  File "/usr/local/lib/python3.7/site-packages/yeelightbt/lamp.py", line 35, in _wrap
    withResponse=True)
  File "/usr/local/lib/python3.7/site-packages/bluepy/btle.py", line 200, in write
    return self.peripheral.writeCharacteristic(self.valHandle, val, withResponse)
  File "/usr/local/lib/python3.7/site-packages/bluepy/btle.py", line 542, in writeCharacteristic
    self._writeCmd("%s %X %s\n" % (cmd, handle, binascii.b2a_hex(val).decode('utf-8')))
  File "/usr/local/lib/python3.7/site-packages/bluepy/btle.py", line 302, in _writeCmd
    raise BTLEInternalError("Helper not started (did you call connect()?)")
bluepy.btle.BTLEInternalError: Helper not started (did you call connect()?)

rytilahti, 2020-04-02:
Is it working when using `yeelightbt` console tool? That error comes from bluepy, so I don't think there is anything we could do inside this project to avoid that.

alexvenom, 2020-04-09:
I am having practically the same error.
The yeelightbt command in the console works fine. I can turn on and off the candelas:

`bash-5.0# yeelightbt --mac f8:24:41:XX:YY:ZZ on
INFO:yeelightbt.connection:Requesting characteristics for uuid 8f65073d-1234-4aaa-afea-XXYYZZ
INFO:yeelightbt.connection:Requesting characteristics for uuid aa7d3f34-1234-41e0-807f-XXYYZZ
We are paired.
Got notif: <Lamp f8:24:41:XX:YY:ZZ is_on(False) mode(20) rgb((2, 7, 8, 0)) brightness(0) colortemp(0)>
Got notif: <Lamp f8:24:41:XX:YY:ZZ is_on(True) mode(20) rgb((2, 7, 8, 0)) brightness(0) colortemp(0)>`

Here is the error:

`Logger: homeassistant.helpers.entity
Source: custom_components/yeelight_bt/light.py:164
First occurred: 9:21:04 AM (19 occurrences)
Last logged: 9:30:22 AM

Update for light.bedside fails
Traceback (most recent call last):
  File "/usr/src/homeassistant/homeassistant/helpers/entity.py", line 279, in async_update_ha_state
    await self.async_device_update()
  File "/usr/src/homeassistant/homeassistant/helpers/entity.py", line 476, in async_device_update
    await self.hass.async_add_executor_job(self.update)
  File "/usr/local/lib/python3.7/concurrent/futures/thread.py", line 57, in run
    result = self.fn(*self.args, **self.kwargs)
  File "/config/custom_components/yeelight_bt/light.py", line 164, in update
    self._dev.state()
  File "/usr/local/lib/python3.7/site-packages/yeelightbt/lamp.py", line 35, in _wrap
    withResponse=True)
  File "/usr/local/lib/python3.7/site-packages/bluepy/btle.py", line 200, in write
    return self.peripheral.writeCharacteristic(self.valHandle, val, withResponse)
  File "/usr/local/lib/python3.7/site-packages/bluepy/btle.py", line 542, in writeCharacteristic
    self._writeCmd("%s %X %s\n" % (cmd, handle, binascii.b2a_hex(val).decode('utf-8')))
  File "/usr/local/lib/python3.7/site-packages/bluepy/btle.py", line 302, in _writeCmd
    raise BTLEInternalError("Helper not started (did you call connect()?)")
bluepy.btle.BTLEInternalError: Helper not started (did you call connect()?)`

rytilahti, 2020-04-10:
You can enable debug logging by manually modifying the `bluepy/btle.py` file (there's a `Debugging` flag) to see if it gives some more output why the helper is not started. My guess is, that this might be a permissions problem, but that's just a guess... The helper itself is located in the same directory that this btle.py file and it's called `bluepy-helper`.